wimpysworld/desktopify #13: Suggestion: Conditioning as a preinstalled image

lah7, 2020-05-05:
Currently the tool allows a user to start with base Ubuntu Server and add a desktop of choice on top. Though, like with previous Ubuntu MATE releases, there is demand for a 20.04 "snapshot" image so users can quickly download once and be ready to go.

:chart_with_upwards_trend:  [Ubuntu MATE is also the **first** third party OS link on the Raspberry Pi downloads page!](https://www.raspberrypi.org/downloads/)

The easiest approach I can think of would be to:

1. Run `desktopify` configured with OEM.
1. **Configure the system to enlarge the root partition on next boot.**
1. Shut down and place the microSD card into the computer.
1. Shrink the partitions to a reasonable size (2-4 GB?)
1. Create a new compressed image with the boot and root partitions.
1. Ship 'em!

:bulb: Could the power of QEMU make it possible to automate the process? I haven't tested this, but may be of interest:

* https://github.com/wimvanderbauwhede/limited-systems/wiki/Raspbian-%22stretch%22-for-Raspberry-Pi-3-on-QEMU

boschkundendienst, 2020-06-15:
You could just backup the desktopyfied version with tar.gz over SSH to a remote box as described [here](https://github.com/boschkundendienst/raspisignage#backup-entire-system) and then deploy it similar to the way archlinux|arm is deployed as described [here](https://archlinuxarm.org/platforms/armv8/broadcom/raspberry-pi-4) using your own .tar.gz. Why always transfer so much zeros using dd?

flexiondotorg, 2020-08-10:
Ubuntu MATE 20.04 beta 1 images are now available for download:

  * https://ubuntu-mate.org/ports/raspberry-pi/

closed 2020-08-10 by flexiondotorg

gilbertohasnofb, 2020-08-26:
> Ubuntu MATE 20.04 beta 1 images are now available for download:

@flexiondotorg This is really great, thanks a lot. It makes installation so much easier when you can just flash an image to an SD card. Do you have any plans to also offer a BerryBoot image (which is in Squash FS format)?