wimpysworld/desktopify #19: cloud-init error when installing kubuntu

zjrohrbach, 2020-05-31:
This is a great idea!  I came across this on the ubuntu podcast.  But I'm having trouble getting it to work in installing kubuntu.

This is my first ever bug report... Not a programmer and don't know if I'm doing it right, but I'm trying to be helpful! 

**Environment:** Raspberry Pi 4 (2GB of RAM), Fresh install of 64-bit 20.04 Ubuntu Server

**Steps to reproduce:** Fresh install Ubuntu Server, `apt update`, `apt upgrade`, `git clone [desktopify-url]`, `desktopify/desktopify --de kubuntu`, reboot

[I get a similar result after the second reboot when using `--oem.`  The first reboot gets me successfully to a config screen; the second reboot has the same error]

**Expected result:** Kubuntu Desktop

**Actual result:** gets hung up at Cloud-init and stops booting

`Cloud-init v. 20.1-10-g71af48df-0ubuntu5 running 'modules:config' at Sun, 31 May 2020 18:49:40 +0000. Up 27.85 seconds.`
`Cloud-init v. 20.1-10-g71af48df-0ubuntu5 running 'modules:final' at Sun, 31 May 2020 18:49:40 +0000. Up 29.40 seconds.`
`Cloud-init v. 20.1-10-g71af48df-0ubuntu5 finished at Sun, 31 May 2020 18:49:40 +0000.  DataSource DataSourceNoCloud [seed=/dev/mmcblk0p1][dsmode=net]. Up 29.63 seconds.`
Then nothing...  

After a hard reset, same result

**Attempts to fix:**

I don't really understand what Cloud-init does, so I'm pretty much flailing around here, but this is what I've tried

Following [this bug](https://bugs.launchpad.net/ubuntu/+source/cloud-init/+bug/1683974), I SSH'd into the stalled system and added `After=systemd-tmpfiles-setup-dev.service` to `/lib/systemd/system/cloud-init-local.service`.  No luck

Following [this](https://raspberrypi.stackexchange.com/questions/107450/ubuntu-server-on-rpi-4-boot-hangs-on-cloud-init), I ssh'd in and did `sudo touch /etc/cloud/cloud-init.disabled`.   After reboot, I get to this point:

`udiskss2.service`
`NetworkManager-wait-online.service`
`cron.service`
`whoopsie.service`
`atd.service`
`blk-availability.service`
`systemd-user-sessions.service`
`serial-getty@ttySO.service`
and then nothing.

zjrohrbach, 2020-05-31:
Update: this appears to be similar to [this bug](https://bugs.launchpad.net/ubuntu/+source/livecd-rootfs/+bug/1860046), but the solution suggested is already implemented in the systemd configuration.

JonTheNiceGuy, 2020-06-01:
As far as I can tell, cloud-init has completed successfully, so it's now whatever comes *after* cloud-init that's hanging...

zjrohrbach, 2020-06-01:
@JonTheNiceGuy : I think you're right after looking more closely.  I just broke my system so I can't reproduce it right now, but it seems possible that there is an issue with the Pi's ethernet. [One of the boot logs seemed to suggest this, but I didn't write it down before I broke the system.]  I'm going to try to get a fresh install set up on WiFi before using desktopify and see if that helps.

Also, I have now reproduced the same issue using `--de lubunutu` instead of `--de kubuntu`.  So I either have a hardware problem, or this is a larger bug.  I might try `--de ubuntu-mate` next to see if it works.

monetaryabyss, 2020-06-01:
The issue above seems to suggest an issue with SDDM. Both Lubuntu and Kubuntu use SDDM as the display manager.

dumser, 2020-06-18:
I haven't attempted troubleshooting yet, but on a RPi4-8GB running Ubuntu 20.04 64-bit, Desktopify --de kubuntu. On boot, I get
```
NetworkManager-wait-online.service
cron.service
whoopsie.service
atd.service
blk-availability.service
systemd-user-sessions.service
serial-getty@ttyS0.service
[   26.x] cloud-init[x]: Cloud-init v. x running 'modules:config' at ...
[   27.x] cloud-init[x]: Cloud-init v. x running 'modules:final' at ...
[   27.x] cloud-init[x]: Cloud-init v. x finished at ... Datasource DataSourceNoCloud [seed=/dev/mmcblk0p1][dsmode=net]. Up 27.x seconds
```
at which point it just sits there. I have ssh'd in to reboot, but I get the same result.

zjrohrbach, 2020-06-19:
Yes, that is precisely my result.  I did get `--de ubuntu-mate` to work.  I'm glad you were able to reproduce on an 8-GB Pi because I was wondering if my 2 GB just had too little RAM.

I also have been meaning to try commenting out the part of the script that switches over to Network Manager given that one of the boot logs suggested an ethernet error.  Another thing I was thinking of trying is just doing `apt install kubuntu` on a fresh installation without `desktopify` to see if it's just an issue with the desktop.

dumser, 2020-06-19:
I ran `Desktopify --de ubuntu-mate` (via ssh) over top of (after) the kubuntu install. During the installation, it asked if I wanted to use lightdm or sddm. I picked lightdm, and it works. I can switch between MATE and Plasma on login -- and other than only allowing 1920x1280 at "0Hz" on my 4k monitor, it appears everything is fine.

flexiondotorg, 2020-08-10:
Pull requests welcome to improve the Kubuntu support :+1:

wxl, 2021-01-08:
AFAIK the fix here is to explicitly install SDDM with the desktop package. I don't have the hardware, so I need someone else to confirm. To be clear, these would be the two changes:

```diff
-    DESKTOP_PACKAGES="kubuntu-desktop"
+    DESKTOP_PACKAGES="kubuntu-desktop sddm"
```

```diff
-    DESKTOP_PACKAGES="lubuntu-desktop"
+    DESKTOP_PACKAGES="lubuntu-desktop sddm"
```

One could easily make those changes in the script and otherwise run it like normal.

PR #66 (2021-12-03): Add video fix for SDDM to work properly.
This change allows SDDM to display properly and should close #19 and #63 . I have also tested a Mate install and no regression was noted. The [documentation](https://www.raspberrypi.org/documentation/configuration/config-txt/boot.md) also indicates this should be set when using the camera module.
closed 2021-12-03 by flexiondotorg