wimpysworld/desktopify #22: git clone asks for password

siduxdevil, 2020-06-27:
I can git clone desktopify to my workstation without a problem. When I try to git clone from Ubuntu server on a RasPi, I get asked for a password!? All I can give it are my github credentials but then I get: fatal: repository … not found. Can someone please explain?

lah7, 2020-06-27:
You can clone using SSH or HTTPS. Likely you'll want to use HTTPS as this doesn't require you to login:

    git clone https://github.com/wimpysworld/desktopify.git

jtremblant, 2020-06-29:
I tried as instructed above with HTTPS as follows:
git clone https://github.com/wimpysworld/desktopify.git
and still asking for a username and password. Cannot get pass it.
I'm re-flashing to see what happens. This is suppose to be straightforward...

evanmiller29, 2020-06-30:
I got that, but then I restarted my pi and git cloned and it worked @jtremblant

jtremblant, 2020-06-30:
It finally worked for me after reflashing again using etcher this time. Thank's

basem95148, 2020-06-30:
Don't use sudo  just use     git clone https://github.com/wimpysworld/desktopify.git
worked for me that way.

orochikun, 2020-07-02:
Same problem , reflashed restarted , same issue 
Also I attempted clone with another github proyect and that worked with no hickups
Any ideas ?

orochikun, 2020-07-02:
Nevermind the url was incorrect TRUST on this make suuuper sure you are typing the correct url

closed 2020-07-05 by flexiondotorg