wimpysworld/desktopify #25: Refuse issue

Dr-Camper, 2020-07-02:
If I run the command "git clone https://github.com/wimpysworld/desktopify.git" there comes the error
"Cloning into 'desktopify'...
fatal: unable to access 'https:// ... ': Failed to connect to github.com port 443: Connection refused"
I'm connected with LAN
I'm from Germany, but why is Github refusing to connect?
Is it my region?

orochikun, 2020-07-02:
There seems to be an error with this project page, since I  attempted to clog from rpi and mac , triple flash , same problem

orochikun, 2020-07-02:
Nevermind,  I had a typo

Dr-Camper, 2020-07-03:
when will the error be fixed

orochikun, 2020-07-03:
Hey , quick suggestion , try to clone other repo from another dev ( using GitHub of course) if that fails then the issue is with GitHub or your connection from Germany to this server, but if that works then .. There might be a typo on your part, trust me I validated 3+times and thought I had it right,

Hope this helps

Dr-Camper, 2020-07-04:
On anothers repo's there comes the same error. I guess my internet provider don't let me access or the router settings are wrong. Thanks anyway for the help.

Dr-Camper, 2020-07-04:
Now i solved the problem. There was an issue in the router settings. Now everthing is working fine.

closed 2020-07-05 by flexiondotorg