wimpysworld/desktopify #31: Ubuntu 64b Gnome Desktop on Rpi 4 graphical issues

shadekh, 2020-07-07:
I just installed the default ubuntu 20.04 64bit Gnome on my rpi 4 8gb via desktopify , and had issues like frequent desktop redraws. I know that Gnome is more recommended, but have been using gnome since long, and I wondered if this is fixable ?

TerraGitHuB, 2020-07-11:
Yes, what it seems to be, is that it's software rendering, so graphic performance doesn't work as it should.

So when they say: "We have hardware acceleration and fakeKMS" It's just software rendering.

Perhaps wait until it's solved.

flexiondotorg, 2020-08-10:
This is now resolved via an updated kernel is available that enables V3D on the Pi 4.

  * If you run `desktopify` today, you'll get the updated kernel.
  * If you have already run `desktopify` then the new kernel is available via `apt` updates.

closed 2020-08-10 by flexiondotorg