wimpysworld/desktopify #34: HDMI audio is not working even after hdmi_drive=2 usercfg

ShashankYadav, 2020-07-09:
Desktopify version :  master branch at [commit](https://github.com/wimpysworld/desktopify/commit/43512fb19e0816a0ae067c8ae66ecd61069200d5)

Issue: HDMI audio is not working.

Extra details: Using Ubuntu 20.04. Also I’m using BerryBoot [(berry boot link)](https://downloads.sourceforge.net/project/berryboot/berryboot-20200612-pi4.zip) to boot Ubuntu [(Ubuntu image link)](http://berrytest.alexgoldcheidt.com:50841/Berryboot_OS_Testing/Raspberry_Pi_4/ubuntu_20.04_preinstalled_server_arm64_rpi4_2020.04.23_berryboot.img.gz) off a usb stick if that is relevant.

Description: I can confirm that /boot/firmware/usercfg.txt has the line `hdmi_drive=2`, which should have enabled the HDMI audio. I’m fine with either HDMI audio or audio from 3.5 mm audio jack so any help is appreciated.

pseudofunizer, 2020-09-16:
Hi there,

In case of my Xubuntu install, I had to setup the audio output to HDMI manually. You can do this by forcing PulseAudio to use HDMI out by default. 

First, you need to check out the available audio sinks.

Open up a local terminal ( `CTRL + ALT + T` -- this cannot be done via SSH) and type in this command:

`pacmd list-sinks | grep -e 'name:' -e 'index:'`

The output _should_ bring out a list like this:

```
    index: 0
        name: <alsa_output.platform-bcm2835_audio.stereo-fallback>
  * index: 1
        name: <alsa_output.platform-bcm2835_audio.stereo-fallback.2>
```

The currently selected/active audio output sink is marked with the asterisk(*). In this example, I have the 3,5mm analog audio output jack selected. The HDMI audio output is index `0`, and its name is `alsa_output.platform-bcm2835_audio.stereo-fallback`.

To swap that one to HDMI, you need to i.e. edit PulseAudio's global default options manually, which I did in my case.

Edit the file `/etc/pulse/default.pa` as `sudo` with your favorite editor, i.e. with the following command:

```sudo nano /etc/pulse/default.pa```

Search through the file for a line like this, should be at the bottom of the file: 

```set-default-sink output```

Usually that is commented out i.e. like this:

```#set-default-sink output```

Uncomment the line and add in the correct index name of your preferred audio sink, in the case of HDMI output, it is:

```set-default-sink output alsa_output.platform-bcm2835_audio.stereo-fallback``` 

Then, try restarting the PulseAudio service.

First kill it with:
```pulseaudio --kill```

Then check if it's still there:
```pulseaudio --check```

Then restart it:
```pulseaudio --start```

Now, run the audio output sink test again:

```pacmd list-sinks | grep -e 'name:' -e 'index:'``` 

If the asterisk has moved from ```index: 1``` to ```index: 0```, you should be able to hear something from the speakers via HDMI now. If not, try i.e. rebooting.

PulseAudio, being what it is, also has a nasty habit of "forgetting" the user's settings. Should PulseAudio forget your preference, see that the line you just added to ```/etc/pulse/default.pa``` is still there. If it is, in some cases the solution (at least for me) was to delete any user settings that may be pulling it back to default analog-out.

```rm -rf ~/.pulse
rm -rf ~/.config/pulse```

Then kill pulseaudio, re-edit ```sudo nano /etc/pulse/default.pa```, add in the line again, save, exit, restart PulseAudio. If I remember correctly, that did the trick for me.

**Speaking of a nice desktop experience...** -- even old versions of Raspbian have a HDMI/analog choice switch that seems to work, unlike the one in Ubuntu. This has been the case for quite a while, I've been running Ubuntu on the RPi4B since v. 18.04 64-bit Server was in unofficial development. My PulseAudio's GUI on the RPi4B doesn't even let me choose any other audio output but the analog one(!). Would be "kinda nice" to have that HDMI/analog selection switch in the panel, but that's just me. Maybe in 2035.

R3Z1H, 2020-09-17:
@pseudofunizer i tried your solution, but it didn't work for me... pulseaudio still defaults back to `fallback.2`
i deleted everything you said.. the modification i made `default.pa` is still there..
but it doesn't work :(

pseudofunizer, 2020-09-17:
> 
> 
> @pseudofunizer i tried your solution, but it didn't work for me... pulseaudio still defaults back to `fallback.2`
> i deleted everything you said.. the modification i made `default.pa` is still there..
> but it doesn't work :(
> 
> Edit: Using Ubuntu-Mate***

@R3Z1H Did you get it to work between reboots though? Also, have you run the latest `sudo apt-get update` and `sudo apt-get upgrade` + perhaps enabled the pre-release (proposed) versions of packages/updates from the repositories? (Software Updater / Software & Updates). 

Also, have you made sure that the HDMI audio output works in i.e. a Raspbian/PiOS ? That could eliminate some possibilities, such as the HDMI cabling or some other component in the chain being unfit for audio transmission...

In any case, does this command show you the two different indexes, one marked with an asterisk(*) like I said in my example?

```pacmd list-sinks | grep -e 'name:' -e 'index:'```

If you see the entries for index 0 and 1 in the list, then at least you've got both the HDMI and the analog listed.

https://askubuntu.com/questions/71863/how-to-change-pulseaudio-sink-with-pacmd-set-default-sink-during-playback

There's a proposal there; 

> To change the default output sink to sink 1 we then run
> 
> ```pacmd set-default-sink 1```
> 
> Sucess can be visualized by opening the Sound Settings menu.

(In your case, it would be ```pacmd set-default-sink 0```.) 

In my Xubuntu install, just by typing in ```pacmd set-default-sink 0``` or ```pacmd set-default-sink 1``` will toggle between the HDMI and analog outputs instantly. Try that command out and then, **without a reboot in between, try playing back audio from i.e. a web browser (don't launch it before you run the ```pacmd```.)**

R3Z1H, 2020-09-17:
@pseudofunizer yes I did everything, except I don't understand what u mean by "enabling prerelease version"...

I downloaded Ubuntu Server image, then ran Wimpy's Desktopify script..

I did update & upgrade everything..

So every time I boot, I have to manually set the proper sound output to be able to hear from HDMI..

As you mentioned this can be done by running `pacmd set-default-sink 0`.. so technically I can just set a crontab @reboot to run that command.. and it will be as if my audio is defaulted to sink 0..

NOTE: just tried using crontab to run that command, but doesn't work for some reason 🙁

pseudofunizer, 2020-09-17:
> 
> 
> @pseudofunizer yes I did everything, except I don't understand what u mean by "enabling prerelease version"...
> 
> I downloaded Ubuntu Server image, then ran Wimpy's Desktopify script..
> 
> I did update & upgrade everything..
> 
> So every time I boot, I have to manually set the proper sound output to be able to hear from HDMI..
> 
> As you mentioned this can be done by running `pacmd set-default-sink 0`.. so technically I can just set a crontab @reboot to run that command.. and it will be as if my audio is defaulted to sink 0..
> 
> NOTE: just tried using crontab to run that command, but doesn't work for some reason slightly_frowning_face

There's something stuck in between the cogwheels there. When logged in as your typical username, go to your home directory with `cd` and then `cd .config/pulse` to get to PulseAudio's directory. Then type in `ls` to get a view of what's in the folder.

There should be a ```**<numbersequence>**-default-sink``` file there that gets automatically created. If it's there, when you edit it with i.e. `nano`, what does it say? If you have the analog output selected (like I do), the only content of the file is:

```alsa_output.platform-bcm2835_audio.stereo-fallback.2```

Change that to 

```alsa_output.platform-bcm2835_audio.stereo-fallback``` 

and try rebooting?

The rationale here is that stuff under ```/etc/pulse``` applies _system-wide_, while the _per-user_ configuration is at `~/.config/pulse`.

Anyway, try that and see what happens. Cheers.

R3Z1H, 2020-09-18:
@pseudofunizer i did what you said, edit `**<numbersequence>**-default-sink` but it reverts back to `fallback.2` everytime i reboot.

But as you suggested, the solution that is working for me now is to add `pacmd set-default-sink 0` command in `Startup Applications`..

you were right about crontab not working due to PulseAudio resetting the profile after user login.

thanks for the solution..

pseudofunizer, 2020-09-19:
> @pseudofunizer i did what you said, edit `**<numbersequence>**-default-sink` but it reverts back to `fallback.2` everytime i reboot.
>
> But as you suggested, the solution that is working for me now is [...]

Hey, I came up with another possible solution when I was working on a duplicate of the install that I had, and had to configure the HDMI output on that one. This was all I needed on Xubuntu.

Edit the `/etc/pulse/default.pa` file again and make sure these two are at the end:

```set-default-sink alsa_output.platform-bcm2835_audio.stereo-fallback
set-default-sink output alsa_output.platform-bcm2835_audio.stereo-fallback```

The idea there is that you set the `default-sink`  **and** the `default-sink output`  over there in `/etc/pulse/default.pa`. 

That worked for me, after that all users at that endpoint had the HDMI selected as the default. 

Let me know if that was of any help.

Cheers.