wimpysworld/desktopify #35: Ethernet Stops working on 2 GB Raspberry pi 4

surajkhanfau, 2020-07-09:


surajkhanfau, 2020-07-09:
Got it working after putting the link up followed up putting a dhcp request. May be check the script?

xenzirril, 2020-07-20:
Been having a similar issue on an RP3 with 32-bit Ubuntu install.  Once I run Desktopify, let it complete and reboot, the Ethernet Port no longer shows any power or status lights.  I can power down and remove the SD, install one with say Raspian loaded, boot up and the Ethernet works fine.  Swap back to the newly Desktopified Ubuntu SD and the Ethernet is again unresponsive.

Have done a few re-installs trying to figure out the best approach.  May have to try manually editing the wpa-supplicant file with my WiFi details next time because nothing I do ONCE Desktopify has run will allow me to use my Ethernet.

I've read that if a Raspberry Pi begins to suffer Power Issues that the Ethernet and USB Ports are the first to suffer.

I have a Raspberry Pi 4 on the way, so maybe this is just a limitation of The RP3 . . .  ?

Any others with this issue?

flexiondotorg, 2020-08-10:
I can't reproduce this issue. If you have more information then please provide it and we can re-open it :slightly_smiling_face:

closed 2020-08-10 by flexiondotorg