wimpysworld/desktopify #44: No Youtube Video (after today's update)

jrullan, 2020-07-27:
Sorry if this is the wrong place to report this, but I was so happy with using Ubuntu Mate with my RPI4 since a couple of weeks ago as my main desktop computer, but today after applying the updates suggested by the Software Updater now I dont have Youtube video in either Firefox or Chromium. A message pops up in the video window: "If playback doesn't begin shortly, try restarting your device".
[image]

flexiondotorg, 2020-08-10:
I can't reproduce this issue.

jrullan, 2020-09-04:
SOLVED!!!!

On the one hand I'm ashamed of the fix, on the other it bothers me this happens at all. 
Well turns out that Youtube (maybe is youtube's fault) will do this on a Raspberry Pi 4 **If** the audio output is not set to one of the HDMI outputs. By default it appears the "default" selected is the headphones jack??? And this brings another problem, all the outputs are named the same and without knowledge of pulse audio command line tools, a regular user such as myself doesn't have a way to know which one is which, specially in a two monitor setup such as mine:

[image]

To be able to play Youtube you need to select any of the other two which are the two HDMI monitors.

[image]
 
Incidentally, I did notice that the first one says in the bottom selector Headphones. In the second it says Analog which is strange since it is actually a digital signal... unless I'm wrong about that.

Anyway, thats it, Solved!

closed 2020-09-04 by jrullan