wimpysworld/desktopify #54: Add and enable vnc server right out of the box

Samk19, 2020-09-03:
Kindly add vnc server support to be enabled and configured before first reboot 

Helps for people having a headless setup

daPhipz, 2020-09-03:
You can connect to the Raspberry Pi via SSH, if you have a headless setup.
Why do you think an out-of-the-box VNC session is needed before the first reboot? (Take this as a friendly question, please - I am genuinely interested!)

If I were to use my Raspberry Pi via VNC, I'd do the initial OS installation etc via SSH, reboot and then configure VNC.

Samk19, 2020-09-03:
Having trouble configuring vnc after reboot tried but some how it didn't work with ubuntu + desktopify (mate) since then using raspberry Pi os (and I'm am stuck on it because vnc is easy to setup with raspi-config) 

Tried installing raspi-config but that also didn't work on Ubuntu server 

If you have any guide (or link to one) that help

mayk78, 2020-09-03:
> Having trouble configuring vnc after reboot tried but some how it didn't work with ubuntu + desktopify (mate) since then using raspberry Pi os (and I'm am stuck on it because vnc is easy to setup with raspi-config)
> 
> Tried installing raspi-config but that also didn't work on Ubuntu server
> 
> If you have any guide (or link to one) that help me to configure vnc with ubuntu mate (desktopify headless) plez share
> 
> Thankyou
> 
> I am currently connected via SSH only (no monitor sob)

I would suggest installing xrdp and connecting trough your rdp client. This also enables you to do multi session, vnc takes controll of the console.  raspi-config is a raspbian command not implemented in ubuntu.  Just google xrdp ubuntu and you should be fine.

Samk19, 2020-09-03:
Thanks will try 

Also has anyone tried https://linuxhint.com/raspberry_pi_headless_mode_ubuntu/ this method is it working?

Is using command sudo apt-get install -y realvnc-vnc-server realvnc-vnc-viewer plus vncserver auto configure realvnc server? 

 I tried using this https://www.digitalocean.com/community/tutorials/how-to-install-and-configure-vnc-on-ubuntu-18-04 but it didn't work for me

That's why created this thread (/issue) so that desktopify could install and auto configure vnc

pseudofunizer, 2020-09-19:
I have my own setup for Ubuntu on RPi4B to get x11vnc working (and it works quite well over local networks, tbh!) -- the only problem I've encountered is the encryption side, it seems like there's plenty of packages out there that are outdated and/or incompatible with that. Out of all the other VNC's I've tried out, the x11vnc is still the best, but rigging up the encryption and/or SSH tunneling part is where things kinda get wonky. At least the last time I tried it out, SSH wasn't available because something was outdated (as usual), yada yada ... ( https://unix.stackexchange.com/questions/420786/how-do-to-bind-x11vnc-to-localhost-only-and-tunnel-through-ssh ) -- I never got that loop-back working properly and ended up just doing an unencrypted machine-to-machine connection over Ethernet. **A WORD OF CAUTION:** if you use VNC without any kind of encryption, you're essentially gambling with _all_ of your data security. Even if everything is inside your own private (W)LAN.

There's an article on x11vnc here: http://www.karlrunge.com/x11vnc/ 

That being said ... It's still the best-working VNC solution out there out of all the different ones I've tried. At least in Xfce4, it makes it a breeze to use your Raspi4B from another computer. I think I could grab my x11vnc setup scripts from somewhere and post them for people to tweak and/or to laugh at.

Please let me know if you got the SSH tunneling to work in x11vnc, I'm also in the Desktopify/Wimpysworld Discord.

Have a nice Saturday,

Cheers!

pseudofunizer, 2020-09-19:
P.S. I must say I'm a bit amazed in comparison on how VNC server works in Raspbian OS's straight out of the box so well. Did they have some kind of a proprietary deal with that one, or has the open source community got a few shoelaces loose in regards to that?

I mean, all of the VNC solutions that I've tried for Ubuntu, have been a tangled mess. Yep, that's one of those few things that you DO miss from Raspbian -- the over-the-top-super-easy configuration of an encrypted VNC server.

**Insert _ * wink wink, nudge nudge * _ here.**

MilkyDeveloper, 2020-09-29:
As an alternative, ssh -X work's too. VNC would be awesome.

MFYDev, 2020-09-29:
@MilkyDeveloper @Samk19 @pseudofunizer @mayk78 @daPhipz 

About VNC, actually it is not hard to set. Please follow these steps \:

1. Run 

```bash
sudo dpkg --add-architecture armhf
sudo apt update
```

2. Donwload and install the deb package for raspberry from https://www.realvnc.com/en/connect/download/vnc/raspberrypi/

3. Copy the following lib files from the Pi4 firmware repository https://github.com/raspberrypi/firmware/tree/master/opt/vc/lib to `/usr/lib`

> libbcm_host.so
libvcos.so
libmmal.so
libmmal_core.so
libmmal_components.so
libmmal_util.so
libmmal_vc_client.so
libchiq_arm.so
libvcsm.so
libcontainers.so

4. Run

```bash
sudo apt --fix-broken install ./VNC-Server-6.7.2-Linux-ARM.deb
```

5. After install process finished, please start and enable them, the command can be found when you finish installing it.

Hope this reply can help you guys. : )

MilkyDeveloper, 2020-09-29:
Woah, thanks! I might make a PR next weekend.

MFYDev, 2020-09-29:
@MilkyDeveloper You are welcome mate~ Finding out this way to install VNC server on my Raspberry Pi 4B really cost me so much time. So I hope you guys won't cost so much time as me on this thing. Just following these steps will perfectly install VNC at last.

MilkyDeveloper, 2020-09-29:
👆 _**There it is :D**_

Samk19, 2020-09-30:
What I requested was to have pre installed vnc so we can have raspberry Pi headless setup up and running right out of the box 

Or some method like  raspi-config to enable vnc in headless setup

The biggest drawback of using mate is that it needs a display for initial config (or a very long setup for initial config)

Adding vnc (or tightvnc for x64) by default will be alot beneficial for a headless mode setup

mayk78, 2020-09-30:
> Bug from
> "name": "View Issue" }, "description": "View this Issue on GitHub",
> "publisher": { "@type": "Organization", "name": "GitHub", "url":

I believe i did the setup headless. installed ubuntu, ran the desktopify 
script, installed xrdp and was set.. i remember i first choose gnome so 
i fiddled with xsessionrc and xsession to change the desktop for the 
remote sessions .  Not 100% sure, but there should be no direct need for 
hooking up a screen to get it all going..

dsanmiguel, 2020-10-02:
I've been trying to get this working for a while for my headless rpi4 and I believe I have finally gotten it to work. I have tried several solutions including @MFYDev suggestion above. For me, it only led to a working VNC that was unencrypted and susceptible. I found I could achieve the same unencrypted VNC result using [another](http://ubuntuhandbook.org/index.php/2016/07/remote-access-ubuntu-16-04/) method that was much easier to follow BUT either way was still an unsecure VNC connection. 

I believe I have finally found a VNC solution that works and is SECURE. This is the [source](http://www.karlrunge.com/x11vnc/) I used:

Scroll towards the bottom and look for the section titled "Tunnelling x11vnc via SSH"
*this [link](http://www.karlrunge.com/x11vnc/x11vnc_opts.html) is also a useful resource for more x11vnc information

I believe @pseudofunizer was closest to solving this when they said that `x11vnc` seemed to be the best option. I haven't tried `ssvnc` which was also mentioned in the source above so that might be less verbose to connect securely since the line of code needed to connect in Option 2 is a bit long but it works nonetheless and is a secure VNC connection. 

*Just fyi I am using the Windows 10 Ubuntu app for ssh and Windows TightVNC Viewer to connect but any ssh program or VNC viewer should work the same.

**I've found two methods that seem to work, but both begin with the same first two steps:**
1) Setup a password under Ubuntu settings/sharing options for VNC which TightVNC (or any other VNC viewer) will prompt for in order to connect
2) Ssh into your raspberrypi-ubuntu and run
`x11vnc -storepasswd` to set up a password

**Option 1:**
3) Then run
`x11vnc -usepw -create` to initialize x11vnc which uses the password you just created and creates a X server session
It will tell you something like "ubuntu:0" is your VNC which is exactly what you type into Remote Host on TightVNCViewer 
4) Put in the password you created in step 1 and you're connected to a secure VNC!

**Option 2:**
3) Run this line of code which ssh's into your ubuntupi and pipes the command to the pi to connect securely, the links I posted above describe the arguments in this line more thoroughly if you are curious
ssh -t -L 5900:localhost:5901 ubuntuusername@whateveryourlocalIPaddressis 'x11vnc -usepw -create -ncache -localhost -display:0'
4) Connect using TightVNC Viewer above and for Remote Host put in "localhost:5900"
5) Put in the password you created in step 1 and you're connected to a secure VNC!

If you have any trouble with either method let me know and I will do my best to help. I hope I have at least provided some resources that are a helpful starting point. Thank you to everyone in this thread for taking the time to contribute as I wouldn't have been able to get to this point without y'all. 

*Sorry for the long post but I have also set up PubKey Authorization when sshing into the pi for even more security which is another topic entirely but when I do this my final code for Option 2 is then:
ssh -i ./.ssh/id_rsa -t -L 5900:localhost:5901 ubuntuusername@whateveryourlocalIPaddressis 'x11vnc -usepw -create -ncache -localhost -display:0'

Ideally for maximum security in ssh you should create a nonadminstrative user account and in sshd_config Disable root and Password Authentication and enable PubKey Authorization

MFYDev, 2020-10-02:
@dsanmiguel I think you must forget something if you follow the steps of mine. What I suggested can definitely lead to an encrypted VNC, which I am using it too. Please have a check.

dsanmiguel, 2020-10-06:
> @dsanmiguel I think the reason why you got an unencrypted VNC is that you might forgot doing something. What I suggested can definitely lead to an encrypted and secure VNC, cause this is just the way that you install the realVNC. The VNC-server of my Raspberry Pi 4B is also installed by these steps without any errors. Please have a check.
> 
> I can reproduce the install process step by step and I finally assure that the method I suggest does not have any troubles.
> 
> If you need help, you could email to me, I am happy to help.

I have given it another go and have now gotten your method to work securely. Thank you for this as it is much simpler than what I suggested above.

pseudofunizer, 2020-10-14:
> I have given it another go and have now gotten your method to work securely. Thank you for this as it is much simpler than what I suggested above.

Seems like a very intriguing choice, but how about distro cross-compatibility and staying up to date (i.e. in case of security issues and such)? Many of Raspbian's packages are notoriously lagging behind in just about everything, to be brutally honest. VNC is one of those things where you want to keep things as sealed as possible. That's why i.e. I restrict it by default into accepting only local area network connections and so forth. 

The problem with `x11vnc` is, for the most part, the extremely loose-ended documentation and implementation of security layer/encryption usage. I'd wish that there in fact _would_ be a solution like the one Raspbian has with `RealVNC` -- that would also keep itself up to date, should security concerns arise, to add to that.

One minor concern in advance considern @MFYDev 's workaround is sticking incoming .debs from another distro into where they don't belong by default (also, Raspbian armhf =>> Ubuntu 20.04.1LTS aarch64). There's a slight chance of breaking some crucial component of your OS. Add to the fore-mentioned the fact that `armhf` within `aarch64` _shouldn't_ pose a problem in of itself, but there's always that slight chance when you start mixing up `armhf` and `aarch64` stuff, not to mention from a different distribution to another ...