wimpysworld/desktopify #59: Allow desktopify to run on 20.10

samlane-ma, 2020-10-03:
I've been successfully using desktopify to test Ubuntu Budgie on the Ubuntu Server 20.10 64-bit daily image.  Though it does seem to work well, I have to edit the script to allow it to run on groovy.  It would be nice to be able to run it "out of the box" on groovy now that 20.10 has reached beta.

aderbas, 2020-10-05:
For me Ubuntu environment using Ubuntu 20.04.1 LTS 64-bit image is lagging in usability. I'm use a Raspberry Pi 4 8Gb and U3 micro SD card.

error414, 2020-12-02:
Ubuntu 20.04 LTS does not work properly with PI400, I would say new PI400 needs new kernel. It works but many errors in console in after install scripts. It's not possible to create initframs or something like that. I'm not familiar with linux.

flexiondotorg, 2021-12-03:
Closing this as 20.10 is now EOL.

closed 2021-12-03 by flexiondotorg